m5stack/M5EPD #6: Add IsCharging API

wizche, 2020-12-21:
To know if USB is attached

DanielPeinhopf, 2020-12-26:
Would also be great to know if the battery is charging or fully charged. But i think this isnt possible because the NCHRG pin of the SLM6600 isn connected :/

openmotion, 2021-01-20:
hello
do you know how to get the battery level ?

markwinap, 2021-01-20:
Hi

uint32_t vol = M5.getBatteryVoltage();
float vol_f = vol / 1000.0f;
sprintf(buf, "%.2f V", vol_f);

markwinap, 2021-01-21:
Also if you want raw values from the ADC

#include "M5EPD.h"
#include "driver/gpio.h"
#include "driver/adc.h"

#define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
#define ADC_FILTER_SAMPLE 8

uint32_t M5EPD::getBatteryVoltage()
{
    uint32_t adc_raw_value = 0;
    for (uint16_t i = 0; i < ADC_FILTER_SAMPLE; i++)
    {
        adc_raw_value += adc1_get_raw(BAT_ADC_CHANNEL);
    }

    adc_raw_value = adc_raw_value / ADC_FILTER_SAMPLE;
    uint32_t voltage = (uint32_t)(esp_adc_cal_raw_to_voltage(adc_raw_value, _adc_chars) / SCALE);
    return voltage;
}

openmotion, 2021-01-21:
> Also if you want raw values from the ADC
>
> #include "M5EPD.h"
> #include "driver/gpio.h"
> #include "driver/adc.h"
>
> #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
> #define ADC_FILTER_SAMPLE 8
>
> uint32_t [...]

hello thanks i have the error : SCALE' was not declared in this scope

openmotion, 2021-01-21:
> uint32_t vol = M5.getBatteryVoltage();
> float vol_f = vol / 1000.0f;
> sprintf(buf, "%.2f V", vol_f);

hello the value returned is : 4.22
is it the battery level ? : 4.22% / 100%

fonix232, 2021-01-22:
@openmotion as the method name suggests, that's the battery VOLTAGE. You need to calculate the percentage yourself, based on the maximum and minimum usable voltages of this battery (this is a LiPo cell, so it would be ~3.2V for minimum, ~4.2V for maximum - yes, I know the official documents and even the sticker on the back says 4.35V, however the charger circuit cuts off when it reaches approx. 4.25V while charging, which translates to around 4.15-4.18V while in use). So you can calculate that 4.22V is approximately:

```
4.25V - 3.2V = 1.05V // This is the range of the percentages, 3.2V being 0%, 4.25V being 100%
4.22V - 3.2V = 1.02V // This is the current voltage on the range
percentage = (1.02V / 1.05V) * 100
percentage = ~97%
```

openmotion, 2021-01-24:
ok is it the right code ?

` uint32_t vol = M5.getBatteryVoltage();
    float vol_f = vol / 1000.0f;
    float xx = vol_f - 3.2;
    float percentage = (xx / 1.05) * 100;
 
    Serial.print("[HTTP] getBatteryVoltage "+String(vol_f)+" \n");
    Serial.print("[HTTP] percentage "+String(percentage)+" \n");`