m5stack/M5EPD #9: Powering on after calling shutdown()

uolot, 2021-01-01:
I struggle with powering the M5Paper after turning it off by executing `shutdown()` with no delay nor date/time.

Here's the code I'm running:

```cpp
void setup()
{
    M5.begin();
    M5.EPD.SetRotation(90);
    M5.EPD.Clear(true);
    M5.RTC.begin();
    M5.BatteryADCBegin();

    canvas.createCanvas(WIDTH, HEIGHT);

    render(canvas, date, tasks, notes, getBatteryLevel());

    M5.shutdown();
}

void loop()
{
}
```

`render()` just draws on the canvas and pushes it out.

After the final `shutdown()` is called, I can see the display as normal and it uses no power at all. However I can't manage to reboot the device. The reset button on the back doesn't seem to be doing anything, and after pressing the button on the side of the device for two seconds, the screen goes blank and stays in that state. The device only resets after connecting/disconnecting from USB-C power source.

What do I miss to make it possible to boot the device up after shutting it down?
Thanks!

felmue, 2021-01-01:
Hi @uolot 

from the M5Paper schematics there should be no difference about whether RTC (timer or alarm) or the power button is powering on the device. (I cannot test that myself as my M5Paper hasn't arrived yet.)

As an experiment, have you tried a timer wakeup using something like `M5.shutdown(5)`? Does that work as expected, i.e. wake up M5Paper after about 5 seconds?

BTW: judging from the picture on the back of the M5Paper the reset button has two functions depending on whether M5Paper is running from battery or USB. (Battery: turn off / USB: reset). So if your M5Paper is running from battery and in shutdown mode (turned off) already, pressing the reset button will do nothing.

Thanks
Felix

dov, 2021-02-17:
I'm having a similar issue for a simple clock application that sleeps most of the time and wakes up once a minute and updates the time. Here are some steps that I did:

1. When the device wakes up I'm checking `M5.BtnP.isPressed()` and using it to distinguish between a manual turn on (the button is pressed) and an RTC wakeup, in which the button is not pressed.
2. I connected a LED to G32 to see when the RTC triggers and the device is turned on.

However, I'm stuck on the issue of how to send drawing commands to the EPD without doing `M5.EPD.Clear(true)`. I.e. I'd like to update only a small region where I'm displaying the clock digits without clearing the whole screen. If I don't do `M5.EPD.Clear()` then there is garbage left in the display. Is there any solution to this?

closed 2024-01-22 by icyqwq