m5stack/M5EPD #13: canvas.createCanvas cannot effect with large area

Daniel1900, 2021-01-11:
canvas.createCanvas(540, 960);
canvas.fillCanvas(15);
canvas.pushCanvas(0,0,UPDATE_MODE_DU4);

cannot work, but change (540, 960) to (540, 440) will effect.

oznotes, 2021-01-21:
Probably creating single canvas with that size is the issue , make it work with multiple canvas 

```
#include <M5EPD.h>

M5EPD_Canvas canvas(&M5.EPD);
M5EPD_Canvas canvas2(&M5.EPD);

void setup()
{
    M5.begin();
    M5.EPD.SetRotation(90);
    M5.EPD.Clear(true);
    canvas.createCanvas(270, 480);
    canvas2.createCanvas(270, 480);
    canvas.setTextSize(3);
    canvas.drawString("Hello World", 45, 350);
    canvas.pushCanvas(0,0,UPDATE_MODE_DU4);
    canvas2.pushCanvas(270,0,UPDATE_MODE_DU4);

}

void loop()
{

}
```

openmotion, 2021-01-21:
you must change the right board, in arduino IDE --> tools -->  board --> m5stackarduino --> your board

closed 2024-01-22 by icyqwq