m5stack/M5EPD #14: Loading a ttf font to the M5paper from the sdcard causes error "but zero-sized"

dov, 2021-01-11:
What I did:

1. Mounted a 16GB sd card to my desktop 
2. Created the directory `/ttf` in the root directory
3. Copied over the font DejaVuSans.ttf to the sdcard to the directory `/ttf`
4. Wrote an arduino sketch with the following code snippet in `setup()`

```
esp_err_t ret = cnv.loadFont("/ttf/DejaVuSans.ttf", 
```
5. Compiled uploaded, opened serial monitor.
6. Result: Get the following message in the serial monitor:

```
FT_Stream_Open: opened `/ttf/DejaVuSerif.ttf' but zero-sized
```

m5EPD version: 0.1.0

eggfly, 2023-10-24:
Same error

dov, 2023-10-24:
It really doesn't seem like anyone cares/is listening...

eggfly, 2023-10-24:
@dov But don't worry maybe I've found why this error appeared. Which file system did you use in tf card?

dov, 2023-10-26:
@eggfly I probably used FAT32, but it's been a couple of years. What did you find out?

icyqwq, 2024-01-22:
This appears to be a file system error. It may be due to a malfunctioning SD card or an incorrect file path, among other potential causes.

dov, 2024-01-22:
@icyqwq How was this completed? What was done?

icyqwq, 2024-01-22:
Try moving the file to the root directory and renaming it to f.ttf. Since the maximum file name length for FS under default settings is 8 bytes, which might be the reason the font file can't be loaded. Additionally, please provide your font file for me to test.

icyqwq, 2024-01-31:
Please reopen this issue if needed.

closed 2024-01-31 by icyqwq