m5stack/M5EPD #17: hello world example

openmotion, 2021-01-20:
hello
if i launch hello world example the screen is blank
the button example is ok only if i press a button
what is the problem ?

oznotes, 2021-01-21:
I dont know the reason behind that maybe creating single canvas with that dimension , maybe EPD driver 
I managed to make it work with creating 2 canvas covering whole screen size 
`#include <M5EPD.h>

M5EPD_Canvas canvas(&M5.EPD);
M5EPD_Canvas canvas2(&M5.EPD);

void setup()
{
    M5.begin();
    M5.EPD.SetRotation(90);
    M5.EPD.Clear(true);
    canvas.createCanvas(270, 480);
    canvas2.createCanvas(270, 480);
    canvas.setTextSize(3);
    canvas.drawString("Hello World", 45, 350);
    canvas.pushCanvas(0,0,UPDATE_MODE_DU4);
    canvas2.pushCanvas(270,0,UPDATE_MODE_DU4);

}

void loop()
{

}`

openmotion, 2021-01-21:
hello 
i found the problem
i didn't selected the good board (esp32-espressif instead of m5paper)

bluejazzCHN, 2021-01-31:
> hello
> i found the problem
> i didn't selected the good board (esp32-espressif instead of m5paper)

if your problem had been resolved , please close this issue. thanks.

closed 2021-01-31 by openmotion