m5stack/M5EPD #18: NFC PN532 grove v1.1

openmotion, 2021-01-23:
hello
i try to connect the nfc PN532 grove v1.1 to my M5paper board. I successfully installed the library and compiled the ReadTag example but i have this error : **Didn’t find PN53x board**
i plugged the NFC module to the port C ( UART )
what’s wrong with my setup ?
thanks

felmue, 2021-01-24:
Hello @openmotion 

have you adapted the ReadTag example to use the proper GPIOs for RX and TX? By default the example uses Serial1 which by default uses GPIO3 and GPIO1 for RX and TX. However on M5Paper different GPIOs are used on port C (UART). Check the sticker on the back of your M5Paper.

Good luck!

Thanks
Felix

openmotion, 2021-01-24:
on the sticker i have : G, 5V TXT, G18, G19
how can i adapt the code ?

`#include <NfcAdapter.h>
#include <PN532/PN532/PN532.h>
#include <M5EPD.h>
#include "driver/gpio.h"
#include "driver/adc.h"

    #include <PN532/PN532_HSU/PN532_HSU.h>

 //HardwareSerial Serial1(1); 
    
    PN532_HSU pn532hsu(Serial1);
    NfcAdapter nfc(pn532hsu);
 
void setup(void) {
    M5.begin();
    Serial1.println("NDEF begin");
    delay(5000);
    Serial1.begin(9600);

    //Wire.begin(18, 19);
    
    Serial1.println("NDEF Reader");
    nfc.begin();
}

void loop(void) {
    SERIAL.println("\nScan a NFC tag\n");
    if (nfc.tagPresent()) {
        NfcTag tag = nfc.read();
        tag.print();
    }
    //delay(5000);
     M5.update();
    delay(100);
}`

openmotion, 2021-01-24:
another test

`#include <NfcAdapter.h>
#include <PN532/PN532/PN532.h>
#include <M5EPD.h>
 
#include <PN532/PN532_SWHSU/PN532_SWHSU.h>
#include "SoftwareSerial.h"
 //HardwareSerial Serial1(1); 
    SoftwareSerial SWSerial(19, 18);
PN532_SWHSU pn532swhsu(SWSerial);
NfcAdapter nfc(pn532swhsu);
 
void setup(void) {
    M5.begin();
    Serial1.println("NDEF begin");
    //delay(5000);
    Serial1.begin(9600);
    Serial1.println("NDEF Reader");
    nfc.begin();
}

void loop(void) {
    SERIAL.println("\nScan a NFC tag\n");
    if (nfc.tagPresent()) {
        NfcTag tag = nfc.read();
        tag.print();
    }
    //delay(5000);
     M5.update();
    delay(100);
}`

felmue, 2021-01-24:
Hello @openmotion 

sorry, I am not sure myself how this can be achieved with the NFC library. But I think you'll find a hint [here](https://wiki.seeedstudio.com/Grove_NFC/).

Check out section "Step 5. Modify the code as below to enable UART communication." but try SoftwareSerial with the correct GPIOs.

Sorry I am not more of a help - but I lack the necessary hardware to try it out myself.

Good luck!
Felix

openmotion, 2021-01-24:
thanks @felmue 

i tried this method but no luck

`#include <NfcAdapter.h>
#include <PN532/PN532/PN532.h>
#include <M5EPD.h>
 
#include <PN532/PN532_SWHSU/PN532_SWHSU.h>
#include "SoftwareSerial.h"
 //HardwareSerial Serial1(1); 

#define RX_PIN 18
#define TX_PIN 19
 #define BAUD_RATE 57600
    SoftwareSerial SWSerial(18, 19);
PN532_SWHSU pn532swhsu(SWSerial);
NfcAdapter nfc(pn532swhsu);
 
void setup(void) {
    M5.begin();
    Serial1.println("NDEF begin");
    //delay(5000);
    Serial1.begin(9600, SWSERIAL_8N1,RX_PIN, TX_PIN);
//    Serial1.begin(BAUD_RATE, RX_PIN, TX_PIN, SWSERIAL_8N1, false, 95, 11);
    Serial1.println("NDEF Reader");
    nfc.begin();
}

void loop(void) {
    SERIAL.println("\nScan a NFC tag\n");
    if (nfc.tagPresent()) {
        NfcTag tag = nfc.read();
        tag.print();
    }
    //delay(5000);
     M5.update();
    delay(100);
}`

closed 2024-01-22 by icyqwq