m5stack/M5EPD #21: Software license

estshorter, 2021-01-30:
I think `src/utility/SHT3x.h` and `SHT3x.cpp` are derived from https://github.com/Risele/SHT3x which is licensed under the GPL-3.0.
If I understand the GPL correctly, this M5EPD repository must be licensed  under the GPL-3.0 too.

lienbacher, 2021-02-22:
I also really like M5Stack Hardware, but M5Stack Software is quite underwhelming so far tbh. 
Please please up your software game m5stack!

estshorter, 2021-02-27:
@Gitshaoxiang ping

ikedam, 2021-05-03:
I ran `grep -ri license .` and found this library contains following licensed codes:

* FreeType license: https://gitlab.freedesktop.org/freetype/freetype/-/blob/master/docs/FTL.TXT
    * src/base/*
    * src/freetype/*
    * src/ft2build.h
    * src/sfnt/*
    * src/smooth/*
    * src/truetype/*
* MIT / BSD / FreeBSD license: https://github.com/Bodmer/TFT_eSPI/blob/master/license.txt
    * [`TFT_eSPI`](https://github.com/m5stack/M5EPD/blob/0.1.1/src/utility/In_eSPI.h#L993), [`Sprite`](https://github.com/m5stack/M5EPD/blob/0.1.1/src/utility/Sprite.cpp#L6).
    * **This repository doesn't look contain the license notice for those files.** Written as "see license in root directory" in source codes, but the license file in the root directory differs the original one.
    * Files in src/Fonts/* look also come from the same source.
* BSD license: https://github.com/m5stack/M5EPD/blob/0.1.1/src/Fonts/GFXFF/license.txt
    * Those files look parts of `TFT_eSPI`.
* 3-clause BSD license: https://github.com/m5stack/M5EPD/blob/0.1.1/src/Fonts/GFXFF/TomThumb.h#L2
    * Tom Thumb 3x5 font.
* MIT license: https://github.com/espressif/WROVER_KIT_LCD/blob/master/README.md
    * [`M5EPD_Canvas`](https://github.com/m5stack/M5EPD/blob/0.1.1/src/M5EPD_Canvas.cpp#L852)
    * Not so sure, but supposed so.
    * **The copyright notice in that file looks broken.** (No copyright holder)
* MIT license: https://github.com/kikuchan/pngle/blob/master/LICENSE
    * [`pngle`](https://github.com/m5stack/M5EPD/blob/0.1.1/src/utility/pngle.h)
* MIT license: https://github.com/ricmoo/QRCode/blob/master/LICENSE.txt
    * [`qrcode`](https://github.com/m5stack/M5EPD/blob/0.1.1/src/utility/qrcode.h)

As far as I understand above licenses (but not so sure):

* Missing copyright and license notices must be added to the repository.
* Software using this library must have copyright notices for above licenses if they are distributed only in binary form.
* Above licenses doesn't prevent this repository licensed under the MIT license, as copyright notices in sources are reachable from anyone.
    * Though, it would be better to have `LICENSE` clarify the licenses this library rely on.
    * Of course, licenses described in the description of this issue prevents this repository licensed under the MIT license.

Gitshaoxiang, 2021-05-06:
thank you feedback. we are working to solve this problem now.

closed 2023-07-16 by Tinyu-Zhao