m5stack/M5EPD #25: Image2gray.py alternative version with more "features"

Crinst, 2021-02-07:
Please add this as alternative to your image2gray.py Python script as it is not very good (at least I had to rewrite half of it).
It adds option to select image_folder path, add black borders around images, creates single image=single file or ImageResource.h with all images it finds.
[images2gray.zip](https://github.com/m5stack/M5EPD/files/5938852/images2gray.zip)

dov, 2021-04-10:
Why don't you add this as a pull request so that it is readable without having to download and open a zip file? And even then, you can't see the diff vs the current contents.

Crinst, 2021-04-10:
I did it this way mainly because of laziness and lack of time. Sorry about that. I will do it standard way (pull request).

MaxSMoke777, 2021-05-09:
Please write a doc for this.

closed 2024-01-22 by icyqwq