m5stack/M5EPD #26: BM8563 wakeup from time

RobinsonLaw, 2021-02-09:
I have test the following cannot wakeup from specific time

        wakeuptime.hour = 10;
        wakeuptime.min = 34;
        wakeuptime.sec = 00;
        M5.shutdown(wakeuptime);

estshorter, 2021-02-09:
A fix similar to [this pull request](https://github.com/m5stack/M5-CoreInk/pull/3) may be helpful.

cat-in-136, 2022-05-06:
The following workaround works without changing the M5EPD library. Same as `M5.RTC.setAlarmIRQ()`
```c
M5.shutdown(RTC_Date(-1, -1, -1, -1), wakeuptime);
```

Other notes:
* Execute `M5.RTC.begin()` in advance
* Pull the USB type-C cable from M5Paper, i.e, do not charge power.
* Use `M5.shutdown()` (or `M5.disableMainPower()`), not deep sleep. See also: https://community.m5stack.com/topic/2892/m5paper-shutdown-deep-sleep-wakeup

Tinyu-Zhao, 2023-07-16:
RTC wake-up timer can only be accurate to the day, hour and minute, but not to the second.

closed 2023-07-16 by Tinyu-Zhao