m5stack/M5EPD #31: the shutdown() function isn't working for me.

Bande11a, 2021-03-05:
Im using PlatformIO to work on a project with the M5 Paper and whenever I call the shutdown() function it just skips it as if it wasn't there.

As PlatformIO doesn't have M5Paper as a board option I'm using M5Stack-Core2 as my board.

Hope someone can help :)

uolot, 2021-03-05:
What are the symptoms of `shutdown()` not working? For me it keeps the display on (because it's e-paper it doesn't consume energy), but the rest of the system shuts down (or goes to deep sleep, not sure) and the battery lasts for weeks if not months.

I'm using `M5Stask-Fire` for the board, here's my Platformio config that works with no issues whatsoever:

```ini
[env:m5stack-fire]
platform = espressif32
board = m5stack-fire
framework = arduino
upload_port = /dev/ttyUSB0
upload_speed = 2000000
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
board_build.partitions = default_16MB.csv
build_flags =
	-DCORE_DEBUG_LEVEL=4
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
lib_deps =
	https://github.com/m5stack/M5EPD.git#main
```

Bande11a, 2021-03-05:
Thanks for the config, and it just skips over it as if it was never there

Bande11a, 2021-03-05:
Okay Ive worked it out. The m5 paper can't shut down if the power cable is pluggin in @uolot thanks for the help anyway

closed 2021-03-05 by Bande11a