m5stack/M5EPD #33: Confusion about power modes

RobertSmart, 2021-03-15:
Hi,

I want to use this device to get data from a server every 10 minutes update the display and then sleep.

This is all working fine, but I need to be able to put the device in config mode to set up the device connections etc.

What I ideally want to do is to have a user press the button on the side of the device to wake it up in config mode, but I can't figure out how the functionality of the device is supposed to work. The sleep seems to behave differently depending on if its plugged in or not. There isn't a way to detect if its main power or not (except I've noticed the battery voltage seems to read 284 when its plugged in?)

should the button on the side (the jog button) wake the device up even if its put to sleep on a timer?

sometimes I find that it just seems to get stuck, like I've somehow turned it off using this button. is it possible to turn it off using this button?

Many thanks,

Rob Smart

felmue, 2021-03-16:
Hello @RobertSmart 

I think the confusion stems from what exactly you mean by sleep. I prefer to distinguish between the two ESP32 sleep modes (light sleep and deep sleep) and the device being in shutdown mode (in which everything is powered off except for the RTC).

For the ESP32 being in either light or deep sleep it still needs to be powered, else it never wakes up.

When the device is shutdown mode (which I have a hunch is what you mean by sleep) only RTC or the power button can wake it up, **but** that is only true when running from battery. If USB is plugged in the device **cannot** go into shutdown mode - it is on all the time.

When you look at the sticker on the back of M5Paper you'll see something labelled MOS (FET). This is the main power switch and it can be activated by the power button, RTC, USB or the ESP32.

When the device is running from battery and is in shutdown mode and you press the power button that temporarily enables the MOSFET, the ESP32 boots up and then hold the enable for you so you can release the power button. The M5Paper is running until your code calls a shutdown method. (Or the battery runs out.)

However when USB is plugged in the MOSFET is enabled permanently. Or in other words, calling a shutdown method with USB plugged in has no effect - the M5Paper stays on.

So yes, the power button (jog button) can wake M5Paper when it has been put into shutdown mode on a timer, but as stated before that only works if running from battery.

The power button has two functions, it enables the MOSFET and it is also connected to a GPIO. So yes, if the M5Paper is in shutdown mode it will power it on and if M5Paper is running the power button can turn it off **if** (and only then) you have coded that in your application. By default the power button cannot turn off the M5Paper.

The main thing to understand is that testing shutdown mode, e.g. by calling a shutdown method can only work and be tested if M5Paper is **not** plugged into USB.

I hope this clears up some of your questions.

Thanks
Felix

Sarah-C, 2021-08-09:
Thanks for the info Felix!

closed 2024-01-22 by icyqwq