m5stack/M5EPD #47: Mac OS Big Sur serial issue tracked down to USB controller: NOT CP2104 but CH9102

ubidefeo, 2021-11-13:
My unit sourced via BangGood came in a few weeks ago.
I finally had time to try it and since the serial port was listed as `/dev/cu.usbmodemxxxx` I just went on and tried to upload a Sketch.
I got a message `cannot upload stub - write to RAM error 0107...`

I could not really find an answer on Google so I took the unit apart.
At the back it clearly states **CP2104** which is a Silicon Labs USB<>Serial controller working out of the box on many systems.
My unit has a CH9102 which is the poster child of the infamous CH340 so many have had issues with on Mac, Windows and Linux.

I looked that up and could basically not track down the driver from a manufacturer, 
but ended up on this thread which bares no solution (yet still valuable information)
https://community.m5stack.com/topic/2175/new-m5stack-failed-to-write-to-target-ram-result-was-01070000/7

and later on this one
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139#issuecomment-904390716

There's a link to download what might look like a sketchy driver but it does indeed work.
Requires authorising Mac OS to run it from the Privacy panel and has text in Chinese.

It does work, so if anyone ends up here and they're on Big Sur you can probably fix your issue like this.
Still not using the CP2104 and not giving hints about it is sketchier than the driver I installed today.

Please make amends on the README adding the chip and a possible driver download.
I was able to track down a possible page from the manufacturer but the driver for Mac does not seem to exist on their page (being it in Chinese I cannot find it myself) but the user in the issue linked above says they received it from the manufacturer

Peace ✌🏼

markus-seidl, 2021-11-15:
:(

ubidefeo, 2021-11-29:
Update:
The drivers available for CH9102 return an error on install `syses didFinishWithResult (null)`
https://docs.m5stack.com/en/core/m5paper_v1.1

kotelnikov, 2022-04-14:
@ubidefeo Hi, friend!!! Try this driver https://github.com/WCHSoftGroup/ch34xser_macos for my M1 PRO it works like a charm. But when installing carefully read readme, all steps important.

hakeemta, 2023-03-18:
> @ubidefeo Hi, friend!!! Try this driver https://github.com/WCHSoftGroup/ch34xser_macos for my M1 PRO it works like a charm. But when installing carefully read readme, all steps important.

This worked for me. Thank you!

Tinyu-Zhao, 2023-07-15:
There are two kinds of serial chips for paper, please try the driver installation method provided in the [document](https://docs.m5stack.com/en/core/m5paper_v1.1).
[image]

closed 2023-07-15 by Tinyu-Zhao