m5stack/M5EPD #49: M5EPD.cpp error

AndiTut, 2022-01-09:
My used Arduino IDE environment:

Version: 2.0.0-rc3
Date: 2021-12-22T15:46:56.004Z
CLI Version: 0.20.2 [13783819]
Copyright © 2022 Arduino SA

I get various error messages that I can't explain.
I ask for a solution, following the small program and the error messages:

#include <M5EPD.h>

M5EPD_Canvas canvas(&M5.EPD);

void setup()
{
    M5.begin();
    M5.EPD.SetRotation(90);
    M5.EPD.Clear(true);
    M5.RTC.begin();
    canvas.createCanvas(540, 960);
    canvas.setTextSize(3);
    canvas.drawString("Hello World", 45, 350);
    canvas.pushCanvas(0,0,UPDATE_MODE_DU4);
}

void loop()
{

}

IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1/tools/sdk/esp32/include/fb_gfx/include" -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_M5Stack_Paper -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"M5Stack_Paper\"" "-DARDUINO_VARIANT=\"m5stack_paper\"" -DARDUINO_PARTITION_default_16MB -DESP32 -DCORE_DEBUG_LEVEL=0 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -DARDUINO_USB_CDC_ON_BOOT=0 "@C:\\Users\\User\\AppData\\Local\\Temp\\arduino-sketch-3C88A513D33701B45A959B337E7998AF/build_opt.h" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\cores\\esp32" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\variants\\m5stack_paper" "-Ic:\\Users\\User\\Documents\\Arduino\\libraries\\M5EPD\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\Wire\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\SPI\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\FS\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\SD\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\SPIFFS\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\HTTPClient\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\WiFi\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.1\\libraries\\WiFiClientSecure\\src" "c:\\Users\\User\\Documents\\Arduino\\libraries\\M5EPD\\src\\ffsupport.cpp" -o "C:\\Users\\User\\AppData\\Local\\Temp\\arduino-sketch-3C88A513D33701B45A959B337E7998AF\\libraries\\M5EPD\\ffsupport.cpp.o"
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp: In member function 'void M5EPD::begin(bool, bool, bool, bool, bool)':
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\**M5EPD.cpp:66:34: error:** call of overloaded 'begin(int, int, int)' is ambiguous
         Wire.begin(21, 22, 400000);
                                  ^
In file included from c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.h:5,
                 from c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:1:
C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\Wire\src/Wire.h:79:10: note: candidate: 'bool TwoWire::begin(int, int, uint32_t)'
     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
          ^~~~~
C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\Wire\src/Wire.h:80:10: note: candidate: 'bool TwoWire::begin(uint8_t, int, int, uint32_t)'
     bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
          ^~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp: In member function 'void M5EPD::BatteryADCBegin()':
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: error: 'ADC1_GPIO35_CHANNEL' was not declared in this scope
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:90:31: note: in expansion of macro 'BAT_ADC_CHANNEL'
     adc1_config_channel_atten(BAT_ADC_CHANNEL, ADC_ATTEN_DB_11);
                               ^~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: note: suggested alternative: 'ADC1_CHANNEL_7'
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:90:31: note: in expansion of macro 'BAT_ADC_CHANNEL'
     adc1_config_channel_atten(BAT_ADC_CHANNEL, ADC_ATTEN_DB_11);
                               ^~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp: In member function 'uint32_t M5EPD::getBatteryRaw()':
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: error: 'ADC1_GPIO35_CHANNEL' was not declared in this scope
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:100:25: note: in expansion of macro 'BAT_ADC_CHANNEL'
     return adc1_get_raw(BAT_ADC_CHANNEL);
                         ^~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: note: suggested alternative: 'ADC1_CHANNEL_7'
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:100:25: note: in expansion of macro 'BAT_ADC_CHANNEL'
     return adc1_get_raw(BAT_ADC_CHANNEL);
                         ^~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp: In member function 'uint32_t M5EPD::getBatteryVoltage()':
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: error: 'ADC1_GPIO35_CHANNEL' was not declared in this scope
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:111:39: note: in expansion of macro 'BAT_ADC_CHANNEL'
         adc_raw_value += adc1_get_raw(BAT_ADC_CHANNEL);
                                       ^~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: note: suggested alternative: 'ADC1_CHANNEL_7'
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\M5EPD\src\M5EPD.cpp:111:39: note: in expansion of macro 'BAT_ADC_CHANNEL'
         adc_raw_value += adc1_get_raw(BAT_ADC_CHANNEL);
                                       ^~~~~~~~~~~~~~~
Using library M5EPD at version 0.1.0 in folder: C:\Users\User\Documents\Arduino\libraries\M5EPD 
Using library Wire at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\Wire 
Using library SPI at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\SPI 
Using library FS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\FS 
Using library SD at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\SD 
Using library SPIFFS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\SPIFFS 
Using library HTTPClient at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\HTTPClient 
Using library WiFi at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\WiFi 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.1\libraries\WiFiClientSecure 
Compilation **error: exit status 1}**

digi-thomas2003, 2022-01-09:
Hello AndiTut,

take a look at this commit: this would help...
commit [2c69459](https://github.com/m5stack/M5EPD/commit/2c69459385f6864c85a02918c9a6cbd9206a3668) 

Regards
Tom

Yardie-, 2022-01-23:
However there are still other i2c bugs 
looking for lastError 
undefined I2C_ERROR_OK etc..

Tinyu-Zhao, 2023-07-15:
[image]
I just tested it and there is no problem, please use the latest M5Stack board definition and the latest M5EPD library version.

closed 2023-07-15 by Tinyu-Zhao