m5stack/M5EPD #51: Q: What is the correct BMP bitmap format?

codingjoe, 2022-01-15:
Hi,

I am currently working on a map too for the M5 Paper. However, I am struggling to convert the tiles into the right format. It all works using PNG map tiles, but I'd like to preprocess the tiles to BMP to improve rendering performance.

I did look into the `bin2code.py`, the function `M5EPD_Canvas::drawBmpFile` doesn't complain but only renders white (nothing).

I'd really appreciate if someone has a hit and possible improve the documentation, to make it simpler for others.

Thanks!
Joe

codingjoe, 2022-01-15:
So, I figured out, that the BMP needs to be an uncompressed BMP3 with 24bit per pixel. However, 24bit are strangely uncommon. I couldn't find a way using ImageMagick or pillow to convert a bitmap with 24bit colors. Is there any tool I could use? Or is there a way to render 8 bit bitmaps?

icyqwq, 2024-01-22:
`from PIL import Image

def save_as_bmp(image_path, output_path):
    # 打开图片
    img = Image.open(image_path)

    # 确保图片是RGB格式
    if img.mode != 'RGB':
        img = img.convert('RGB')

    # 保存为24位RGB BMP文件
    img.save(output_path, format='BMP', bits=24)

# 使用示例
input_image = 'path_to_your_input_image.jpg'  # 输入图片的路径
output_image = 'output_image.bmp'  # 输出图片的路径

save_as_bmp(input_image, output_image)
`

icyqwq, 2024-01-22:
```
from PIL import Image

def save_as_bmp(image_path, output_path):
    img = Image.open(image_path)

    if img.mode != 'RGB':
        img = img.convert('RGB')

    img.save(output_path, format='BMP', bits=24)

input_image = 'path_to_your_input_image.jpg'  
output_image = 'output_image.bmp' 
save_as_bmp(input_image, output_image)
```

closed 2024-01-22 by icyqwq

codingjoe, 2024-01-22:
Hi @icyqwq,

Thanks for getting back to me. I actually ended up implementing support for 4bit (16 color) bitmaps for my project. If you are interested, I'd be happy to contribute upstream support.
You can find my implementation here https://github.com/codingjoe/M5Maps

Cheers!
Joe