m5stack/M5EPD #53: M5paper stuck in - M5EPD initializing...OK - loop

NilsMinor, 2022-10-30:
### Describe the bug

I tested some examples from this library(Touch, Button ...) and always see the same issue with my M5Paper board. It constantly restarts and is stuck in a loop printing "M5EPD initializing...OK" The full verbose output can be found below:

[  2856][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2957][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  2957][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  2958][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  2967][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3072][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3173][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3274][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  3274][D][GT911.cpp:35] begin(): GT911: Initialization
[  3274][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
OK
[     5][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   453][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

### To reproduce

This happens with every project. I selected the M5paper as board (see image)
[image]

I also tried to upload the example with Platform IO with the following configuration:

[env:next-dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 2000000
monitor_speed = 115200
upload_port = /dev/cu.wchusbserial537A0111951
lib_deps = m5stack/M5EPD@^0.1.4

### Expected behavior

I haven't change a single line of code so just wanted to run the example. The M5Burner works for me, I have tested this :)

### Environment

- OS:
- IDE &IDE Version:
- Tested with Arduino IDE and PIO in Vscode, both in the latest version
- Repository Version:

### Additional context

- Can you help me, why is the board restarting all the time?
- What is the right platform-io configuration?

Thanks and best regards,
Nils

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5EPD/issues?q=)
- [X] My report contains all necessary details

NilsMinor, 2022-10-30:
FYI the exact same behavior happens with this example project (https://github.com/m5stack/M5Paper_FactoryTest )

I get this as output, over and over again:
M5EPD initializing...[  2629][D][M5EPD_Driver.cpp:68] begin(): Init SUCCESS.
[  3256][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x400fc9c8 arg=0x3ffbde2c
[  3402][W][sd_diskio.cpp:186] sdCommand(): token error [0] 0x12
[  3403][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  3403][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  3424][D][GT911.cpp:35] begin(): GT911: Initialization
[  3424][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  3536][D][font_render.c:63] font_cache_init(): Chace use 51104 bytes

Any help is highly appreciated as I really have no idea why it is restarting all the time, but burning the examples from m5burner works fine with my setup (mc+usb cable + m5paper)

NilsMinor, 2022-10-30:
Ok I've made some progress. It seems to be related to the touch initialization and the esp framework version. I used the latest version 5.2.0 and this didn't work BUT with version 5.1.1 of ESP it is working. I have no idea what has changed but this was breaking this library :)

GreenOlvi, 2022-10-30:
Hi, I have the same error and boot loop. Managed to narrow the error down to newer version of [espressif/arduino-esp32](https://github.com/espressif/arduino-esp32); worked on 2.0.4, crashing on 2.0.5.

Managed to get code working by adding
`platform_packages = framework-arduinoespressif32@3.20004.220825`
to `platformio.ini`.

But I also have no idea which change broke it.

NilsMinor, 2022-11-03:
Thanks for the insight :) lets hope it will be fixed in the next release :)

closed 2022-11-03 by NilsMinor