m5stack/M5EPD #60: Unable to build with arduino-esp32 m5paper definition

ZeroChaos-, 2024-09-25:
### Describe the bug

The m5stack board definitions are based on arduino-esp32 2.x but I want some newer wifi stuff so I tried building with the now upstream arduino-esp32 m5paper board target and it fails.

### To reproduce

Select board arduino-esp32 -> m5paper

make a sketch:
```
#include <M5EPD.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Compile fails

### Expected behavior

Compile should work :-)

### Screenshots

```
Compiling sketch...
/home/zero/.arduino15/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-g++ -MMD -c @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/cpp_flags -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_M5STACK_PAPER -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"M5STACK_PAPER\"" "-DARDUINO_VARIANT=\"m5stack_paper\"" -DARDUINO_PARTITION_default_16MB "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:m5stack_paper:PSRAM=enabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=16M,UploadSpeed=1500000,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -DARDUINO_USB_CDC_ON_BOOT=0 @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/defines -I/home/zero/Arduino/sketch_sep25a -iprefix /home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/include/ @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/includes -I/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/qio_qspi/include -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/cores/esp32 -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/variants/m5stack_paper -I/home/zero/Arduino/libraries/M5EPD/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Wire/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPI/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SD/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPIFFS/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/HTTPClient/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Network/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/NetworkClientSecure/src @/tmp/arduino_build_381827/build_opt.h @/tmp/arduino_build_381827/file_opts /tmp/arduino_build_381827/sketch/sketch_sep25a.ino.cpp -o /tmp/arduino_build_381827/sketch/sketch_sep25a.ino.cpp.o
Compiling libraries...
Compiling library "M5EPD"
/home/zero/.arduino15/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-gcc -MMD -c @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/c_flags -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_M5STACK_PAPER -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"M5STACK_PAPER\"" "-DARDUINO_VARIANT=\"m5stack_paper\"" -DARDUINO_PARTITION_default_16MB "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:m5stack_paper:PSRAM=enabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=16M,UploadSpeed=1500000,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -DARDUINO_USB_CDC_ON_BOOT=0 @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/defines -I/home/zero/Arduino/sketch_sep25a -iprefix /home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/include/ @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/includes -I/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/qio_qspi/include -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/cores/esp32 -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/variants/m5stack_paper -I/home/zero/Arduino/libraries/M5EPD/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Wire/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPI/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SD/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPIFFS/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/HTTPClient/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Network/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/NetworkClientSecure/src @/tmp/arduino_build_381827/build_opt.h @/tmp/arduino_build_381827/file_opts /home/zero/Arduino/libraries/M5EPD/src/font_render.c -o /tmp/arduino_build_381827/libraries/M5EPD/font_render.c.o
/home/zero/.arduino15/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-g++ -MMD -c @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/cpp_flags -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_M5STACK_PAPER -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"M5STACK_PAPER\"" "-DARDUINO_VARIANT=\"m5stack_paper\"" -DARDUINO_PARTITION_default_16MB "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:m5stack_paper:PSRAM=enabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=16M,UploadSpeed=1500000,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -DARDUINO_USB_CDC_ON_BOOT=0 @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/defines -I/home/zero/Arduino/sketch_sep25a -iprefix /home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/include/ @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/includes -I/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/qio_qspi/include -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/cores/esp32 -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/variants/m5stack_paper -I/home/zero/Arduino/libraries/M5EPD/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Wire/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPI/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SD/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPIFFS/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/HTTPClient/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Network/src -I/home/zero/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/NetworkClientSecure/src @/tmp/arduino_build_381827/build_opt.h @/tmp/arduino_build_381827/file_opts /home/zero/Arduino/libraries/M5EPD/src/M5EPD_Canvas.cpp -o /tmp/arduino_build_381827/libraries/M5EPD/M5EPD_Canvas.cpp.o
/home/zero/.arduino15/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-g++ -MMD -c @/home/zero/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32/flags/cpp_flags -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_M5STACK_PAPER -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"M5STACK_PAPER\"" "-DARDUINO_VARIANT=\"m5stack_paper\"" -DARDUINO_PARTITION_default_16MB "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:m5stack_paper:PSRAM=enabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=16M,UploadSpeed=1500000,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -mfix-esp32-psram-c [... cut ...]
- IDE &IDE Version: Arduino 1.8.18
- Repository Version: tested with git HEAD 8db3e288b48589c77e52981bbcc3c3b4c443b2c0

### Additional context

I am able to easily test from git or custom patches as needed.  I would be greatful to get this library compatible with a newer version of arduino-esp32 so I can use the newer wifi settings.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5EPD/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-09-29:
Sorry, ours does not support arduino-esp32 2.x yet.

ZeroChaos-, 2024-09-30:
I realize that.  I don't know how much work it will be to fix it, and reporting bugs and testing is about my skill level with this arduino stuff for the moment.  If you want to work on adding support by fixing the reported issue I can test.

Tinyu-Zhao, 2024-09-30:
Sorry for the inconvenience. It will be much more convenient for you to read it in conjunction with this [document](https://github.com/espressif/arduino-esp32/blob/master/docs/en/migration_guides/2.x_to_3.0.rst#ledc).

ZeroChaos-, 2024-09-30:
Thanks.  I am out of my depth here but I can read and learn :-)

ric-tim, 2024-12-13:
Still no solution for this M5EPD library ?

jsoeterbroek, 2025-05-04:
Running into exact same problem:

OS: Linux
IDE &IDE Version: Platformio
platform = espressif32
framework = arduino

remisarrailh, 2025-05-14:
Not tested for now but i'm working on a fix:
https://github.com/remisarrailh/m5epd

For now I have an issue if SD card is enable, screen doesn't update, you can disable it with this
```
M5.begin(true,false,true,true,true);
```

Platformio.ini (i'm using pio-arduino but it should works without it)

```
[env:m5stack_paper]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = m5stack_paper
framework = arduino
lib_deps = https://github.com/remisarrailh/M5EPD
```