m5stack/M5EPD #61: How to build for M5Paper S3?

ipepe, 2025-03-24:
I have Arduino IDE installed and I have been trying to build even a simple demo and uploading it to my M5PaperS3 but there are all sorts of issues. Is there a tutorial or guide for M5PaperS3 specifically?

Tinyu-Zhao, 2025-03-25:
ref https://docs.m5stack.com/en/arduino/m5papers3/program

ipepe, 2025-03-25:
I have managed to compile and upload examples from M5GFX but I cannot build from M5EPD like this one: https://github.com/m5stack/M5EPD/blob/main/examples/Basics/DISPLAY_URL_JPG/DISPLAY_URL_JPG.ino it throws an error with message: 
```
error: 'VSPI' was not declared in this scope; did you mean 'SPI'?
  117 |     M5EPD_Driver(int8_t spi_index = VSPI);
      |                                     ^~~~
      |                                     SPI
```

ipepe, 2025-03-26:
Also is M5PaperS3 factory demo available somewhere?

PaulskPt, 2025-04-09:
I have the same issue as @ipepe when trying to build a simple hello world for the M5PaperS3. Using Arduino IDE v.2.3.5.
```
\...\Arduino\libraries\M5EPD\src/M5EPD_Driver.h:117:37: error: 'VSPI' was not declared in this scope
     M5EPD_Driver(int8_t spi_index = VSPI);
```

Tinyu-Zhao, 2025-04-10:
The content in this repository is only applicable to M5Paper(ESP32), not M5PaperS3 (ESP32S3).

closed 2025-04-10 by Tinyu-Zhao

PaulskPt, 2025-04-10:
OK, thank you.